elementary/settings-bluetooth #3: Default window size doesn't show new Bluetooth devices

elementaryBot, 2017-04-10:
When you open the discover devices tab in switchboard or if you go to it via the bluetooth indicator on wingpanel I get the same problem. The default window is too small so that you can't even see the new devices when they appear; I thought that just meant it wasn't going to find my device but when I went to full screen it showed the device perfectly. 

It seems to just be an issue of proportioning the "panes" in the window a little better.

Launchpad Details: [#LP1665956](https://bugs.launchpad.net/bugs/1665956) Cody - 2017-02-19 03:40:53 +0000

Jimb5, 2017-12-14:
Here is a screenshot for reference when I open the discover new device window
[image]

danrabbit, 2018-03-06:
Fixed since we don't use GNOME bluetooth wizard anymore

closed 2018-03-06 by danrabbit