elementary/settings-bluetooth #5: Bluetooth mouse doesn't connect automatically

elementaryBot, 2017-04-10:
Unlike other distributions (Fedora GNOME and Xubuntu come to mind.)
Some bluetooth devices don't automatically connect.
This is just a wishlist but it's really handy.
But, it's weird because when I turn on my bluetooth mouse and jiggle it around
the bluetooth indicator in the panel flickers between bluetooth connected and not connected.

Launchpad Details: [#LP1439016](https://bugs.launchpad.net/bugs/1439016) Derrick Kinsey - 2015-04-01 06:13:48 +0000

elementaryBot, 2017-04-10:
I, too, am having the same issue, especially keeping bluetooth connection going if inactive or over suspend. Necessitates re-pairing the bluetooth mouse. 

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1439016/comments/1) Brendan Brannigan - 2015-07-03 15:47:32 +0000

elementaryBot, 2017-04-10:
OP, can you still confirm this issue? I am unable to reproduce. Bluetooth mouse is connecting immediately here.

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1439016/comments/2) Daniel Fore - 2016-12-17 17:42:31 +0000

Philip-Scott, 2017-05-11:
Looks like this is responsible by the device's protocol it used. I have 2 bluetooth keyboards, one of them connects automatically when i press any key, but with the other i have to manually click on the indicator's widget.

The Switch's controller needs to both press the pair button and click the indicator's widget for it to pair

cassidyjames, 2017-12-22:
@Philip-Scott but if GNOME and Xubuntu are behaving better and we could behave better, this might still be a valid bug.

cassidyjames, 2017-12-22:
Sounds like a duplicate of #4

peteruithoven, 2018-09-26:
On Juno beta (1) my M720 Triathlon mouse reconnect automatically. 
When I turn my Anker SoundBuds Slim headphones they seem to be connected momentarilly but then disconnect again. I have to manually connect them again. Quite similar to:#48

peteruithoven, 2018-10-11:
Thanks to a great tip I've been able to have my audio Bluetooth devices auto connect as well!

Apparently I needed to trust the devices:
1. Run: `$ bluetoothctl`
2. Run: `paired-devices`
3. Copy the ID of the audio device that doesn't auto connect
4. Run: `trust {id}`

Source: https://wiki.archlinux.org/index.php/Bluetooth_headset#Setting_up_auto_connection

I don't know why my mouse does auto connect, even though according to `bluetoothctl` it's not trusted.