elementary/settings-bluetooth #6: Bluetooth sound device connect to elementary only after sound device restarted.

elementaryBot, 2017-04-10:
After turning on bluetooth elementary just don't make connection with bluetooth sound device. Additional dont have option to turn on bluetooth and let device connect automaticly.

Launchpad Details: [#LP1611917](https://bugs.launchpad.net/bugs/1611917) Sertleris - 2016-08-10 18:56:22 +0000

elementaryBot, 2017-04-10:
It's like bluetooth autoconnect feature available on paired devices is not working. I need to go to my phone and force it to connect manually to Elementary by selecting computer in the bluetooth available paired devices on the phone. It's not an Android issue. My Android device autoconnects to every device (car, speaker) it finds. 

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1611917/comments/1) miguelgilmartinez - 2016-09-06 06:09:58 +0000

elementaryBot, 2017-04-10:
Bluetooth sound device can use High Fidelity Playback only after on/off connection. Every time after device reboot again - again...

+video attachment

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1611917/comments/2) Sertleris - 2016-09-06 19:39:09 +0000

elementaryBot, 2017-04-10:
Any news?

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1611917/comments/3) Sertleris - 2016-10-09 17:06:56 +0000

elementaryBot, 2017-04-10:
Unfortunately, this issue doesn't seem to have gathered volunteer interest. 

If it's really important for you to get this issue fixed, consider placing a bug bounty: https://www.bountysource.com/teams/elementary/bounties

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1611917/comments/4) Zisu Andrei - 2016-10-10 00:10:01 +0000

elementaryBot, 2017-04-10:
I too am experiencing the same issue. Anyone got a better workaround for now? 

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1611917/comments/5) Joshua - 2016-10-28 12:40:51 +0000

elementaryBot, 2017-04-10:
no

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1611917/comments/6) Sertleris - 2016-11-17 21:25:06 +0000

cassidyjames, 2017-12-22:
Might also be a duplicate of #4?

peteruithoven, 2018-09-26:
On Juno beta (1) my M720 Triathlon mouse reconnect automatically when turning bluetooth on. 
It also automatically connects to my Anker SoundBuds Slim headphones, but only momentarily, then disconnect. I have to manually connect them again. Quite similar to:#48 I don't need to restart my headphones.