elementary/settings-bluetooth #7: Missing "send file" after selecting file for bluetooth transfer.

elementaryBot, 2017-04-10:
When sending a file via bluetooth, there is no actual way to send the file, after selecting it.

After the pairing is done, one can select "send files" in the bluetooth settings. This opens a dialogue to select the file(s) to be sent. But from there, there is no way to actually send the selected files. There is no "send file" button, and pressing enter does nothing either.

The file selection also does not have a close button, so the only way to get rid of it, is using the dock.

Launchpad Details: [#LP1629240](https://bugs.launchpad.net/bugs/1629240) Jonas Ehrhard - 2016-09-30 09:32:59 +0000

elementaryBot, 2017-04-10:
right....also i can't transfer file from my tab into the laptop using bluetooth.....but i can send from laptop to tab. :-P

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1629240/comments/2) Obro Ahmed - 2016-09-30 11:10:52 +0000

elementaryBot, 2017-04-10:
This looks a lot like this: http://elementaryos.stackexchange.com/questions/1638/buttons-are-missing-in-the-file-chooser-dialog-what-should-i-do

It's a settings problem. There's nothing that the bluetooth settings pane can do to resolve this

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1629240/comments/3) Daniel Fore - 2016-12-18 03:54:35 +0000

elementaryBot, 2017-04-10:
It may be a settings problem, but the setting mentioned on Stackexchange is set:

````
jonas@jonas-elementary:~$ gsettings get org.gnome.settings-daemon.plugins.xsettings overrides
{'Gtk/DialogsUseHeader': <0>, 'Gtk/ShellShowsAppMenu': <0>, 'Gtk/DecorationLayout': <'close:menu,maximize'>}
````

Also, I have never (knowingly) meddled with my gsettings, so maybe the standard setting is wrong.

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1629240/comments/4) Jonas Ehrhard - 2016-12-18 11:46:34 +0000

elementaryBot, 2017-04-10:
I also have the same problem.  

$ gsettings get org.gnome.settings-daemon.plugins.xsettings overrides
{'Gtk/DialogsUseHeader': <0>, 'Gtk/ShellShowsAppMenu': <0>, 'Gtk/DecorationLayout': <'close:menu,maximize'>}

Launchpad Details: [#LPC](https://bugs.launchpad.net/switchboard-plug-bluetooth/+bug/1629240/comments/5) Sivaraj D - 2017-02-12 05:17:27 +0000

danrabbit, 2017-04-10:
Closing this because this dialog is not part of this codebase

closed 2017-04-10 by danrabbit