elementary/settings-bluetooth #8: Bluetooth audio device is not recognized and does not pair automatically

elementaryBot, 2017-04-10:
I am using a wireless bluetooth speaker (HP Wireless speaker S6500) and I am facing issues in pairing the speaker with elementary OS 0.4 Loki (64 bit). Eventhough the audio device is paired successfully, it does not connect automatically with the system upon restart.

Please see if this issue is reported earlier and provide/ suggest a fix if the issue is found to be true.

Thanks,
Raja Pandian C

Launchpad Details: [#LP1630328](https://bugs.launchpad.net/bugs/1630328) RajaPandian - 2016-10-04 17:31:42 +0000

peteruithoven, 2018-09-26:
On Juno beta (1) my M720 Triathlon mouse reconnect automatically. My Anker SoundBuds Slim headphones don't. This might be related to :#48

leonardo-lemos, 2018-10-11:
This bug is related to #48 and it can supersede this bug. Closing.

closed 2018-10-11 by leonardo-lemos