elementary/settings-bluetooth #15: The "Bluetooth Device Search" window does not open expanded

GabrielMGitHub, 2017-05-03:
The "Bluetooth Device Search" window does not open expanded, every time you have to resize the window.
[image]
[image]
[image]

GabrielMGitHub, 2017-05-03:
The window should be showing this way.
[image]

danrabbit, 2017-05-03:
This dialog actually comes from GNOME and is not part of this code base. It is unfortunately an upstream issue and invalid for this repo

closed 2017-05-03 by danrabbit