elementary/settings-bluetooth #19: Cannot find bluetooth-wizard

mblarsen, 2017-08-26:
Nothing happens when clicking "Discover Devices.."

I ran `switchboard` from the commandline and repeated. This is what's in the log:

```
[WARNING 14:50:31.583131] MainView.vala:122: Failed to execute child process "bluetooth-wizard" (No such file or directory)
```

cassidyjames, 2017-12-22:
@mblarsen are you on elementary OS Loki?

cassidyjames, 2017-12-22:
I am seeing this on my relatively new Artful-based FrankenJuno install as well. It looks like bluetooth-wizard is [provided by unity-control-center](http://manpages.ubuntu.com/manpages/artful/man1/bluetooth-wizard.1.html).

torikulhabib, 2018-02-28:
how about this source mr. group elementary
you redistribute blueman.
blueman working but dbus accepted confirmation not working.
i am sorry, maybe can help you.
https://github.com/blueman-project/blueman

mblarsen, 2018-02-28:
@cassidyjames sorry for the late reply. Yes it is on Loki.

PR #28 (2018-03-06): Show available device in-line, remove the need for another utility
Fixes #19
closed 2018-03-06 by danrabbit