elementary/settings-bluetooth #21: Bluetooth enables again after unlock.

JonnyCodewalker, 2017-10-16:
Currently Bluetooth enables itself again each time you unlock the PC.

Steps to reproduce: 
Turn of Bluetooth -> Lock PC -> Unlock PC

Hardware:
MSI GP60 2PE

Software:
Elementary 4.1

drom98, 2018-08-28:
Still have this issue on Juno.

Gabriel-p, 2018-10-22:
I guess #42 (related to #59) could address this one by disabling Bluetooth completely?

DavidTR49, 2019-04-15:
Still not working, Bluetooth enable itself after unlocking the PC

xana452, 2019-07-09:
Issue is still present, July 9th 2019.

Blast-City, 2019-07-28:
Althought the indicator shows as enabled, the gsetting entry shows as false.

io.elementary.desktop.wingpanel.bluetooth bluetooth-enabled

4jNsY6fCVqZv, 2019-08-14:
Hi, since a Bluetooth plug release is also on your list for the 5.1 release, I want to ask you if you can fix and release the issue before?
It's a very annoying issue for me, because I'm a user who doesn't need Bluetooth at all, but has to switch it off manually all the time. And this several times a day.

Blast-City, 2019-08-14:
> Hi, since a Bluetooth plug release is also on your list for the 5.1 release, I want to ask you if you can fix and release the issue before?
> It's a very annoying issue for me, because I'm a user who [...]

Hi.

Have you tried to disable bluetooth on the login screen? That seems to fix the issue.

4jNsY6fCVqZv, 2019-08-15:
Thanks for the hint! I can try that out.
But it's more a workaround than a real fix, isn't it?

Blast-City, 2019-08-15:
> Thanks for the hint! I can try that out.
> But it's more a workaround than a real fix, isn't it?

Hi.

Yeah. I should have said workaround. Sorry.

4jNsY6fCVqZv, 2020-01-04:
Is there anyone in the community who wants to fix this issue?

4jNsY6fCVqZv, 2020-07-14:
@ryonakano @jeremypw Is that something you can imagine to fix?

jeremypw, 2020-07-18:
@4jNsY6fCVqZv I am not entirely sure this a bleutooth plug issue - I suspect the issue is at a deeper level.  There is a vaguely related Greeter issue - https://github.com/elementary/greeter/issues/168 - which has not yet gained much traction.  Unfortunately, I am not really a systems level programmer so if its not a problem with the bluetooth plug or indicator I may not be able to help much.

One of these days I'll have to try and get my head around system level stuff ;-)

jeremypw, 2020-07-18:
Strangely there is another issue which appears to be the opposite of this one (see above links).

4jNsY6fCVqZv, 2020-07-19:
@jeremypw 

> Unfortunately, I am not really a systems level programmer so if its not a problem with the bluetooth plug or indicator I may not be able to help much.

How can we find out whether it is at the plugin level or at a deeper system level?

Where do you see a parallel to the linked Greeter Issue?

And who in the development team would be a good person to contact if the underlying cause is at a deeper system level?

jeremypw, 2020-07-19:
>How can we find out whether it is at the plugin level or at a deeper system level?

I'll try and exclude the plug/indicator from being responsible if I can (i.e. check they are only responding to info received from the system)

>Where do you see a parallel to the linked Greeter Issue?

They both involve bluetooth and logging out and in.  They may not be related but my gut feeling is that they are.

>And who in the development team would be a good person to contact if the underlying cause is at a deeper system level?

The correct procedure is to raise the issue in the appropriate repo which will be monitored by the appropriate devs.  If it turns out it is not in the correct repo they will move it.

4jNsY6fCVqZv, 2020-07-19:
probably a duplicate. https://github.com/elementary/switchboard-plug-bluetooth/issues/123

whether https://github.com/elementary/switchboard-plug-bluetooth/issues/82 is related to this? after all, it seems as if the bluetooth settings made by the user are not respected as well.

looks like the issue is also addressed in the wingpanel-indicator-bluetooth project - in conjunction with greeter, that it automatically activates or deactivates itself.
https://github.com/elementary/wingpanel-indicator-bluetooth/issues/117
https://github.com/elementary/wingpanel-indicator-bluetooth/issues/23
https://github.com/elementary/wingpanel-indicator-bluetooth/issues/7

> Alternatively raise it in https://github.com/elementary/os/issues which acts as a catch-all.

Cassidy says it shouldn't act as a catch-all repo. https://github.com/elementary/os/issues/381#issuecomment-655647483

jeremypw, 2020-07-20:
>Cassidy says it shouldn't act as a catch-all repo. elementary/os#381 (comment)

Oh, sorry. At one stage there was a repo that was intended for issues that did not clearly fit in another project - I thought it was that one.

4jNsY6fCVqZv, 2020-07-20:
Cassidy's commentary is rather young. Until then I had understood the purpose of the repository as a collection point as well and used it that way, since my concerns didn't fit into existing repositories for me.
I also ask myself what to do with ideas that would require new repositories first, should there ever be a wish to implement them. @cassidyjames 
For example, a cross-system migration and backup solution.

cassidyjames, 2020-07-20:
[elementary/triage](https://github.com/elementary/triage) is the repo for issues that don't have a better home.

4jNsY6fCVqZv, 2020-08-02:
@jeremypw When could you take the time to find a solution to this issue?

jeremypw, 2020-08-03:
@4jNsY6fCVqZv This seems to be fixed in elementary daily - I do not experience it anymore.  I think it was fixed by https://github.com/elementary/wingpanel-indicator-session/pull/126

BugFixesAndPerformanceImprovements, 2024-12-04:
This seems to be back with elementaryOS 8.0