elementary/settings-bluetooth #23: Crash switchboard if wingpanel-indicator-bluetooth is not installed

quequotion, 2018-01-10:
https://github.com/elementary/switchboard-plug-bluetooth/blob/a1000f53c06143c4c736793756fc396c16242809/src/Services/Manager.vala#L51

If one attempts to start switchboard with the bluetooth plugin installed, but not the indicator, switchboard will crash.

`[GLib-GIO] Settings schema 'org.pantheon.desktop.wingpanel.indicators.bluetooth' is not installed`

One way to resolve this would be to have the switchboard plug depend on the indicator in distributor's packaging. I'm not sure how likely it is that someone might want to use switchboard's plug with another indicator, but it is a possibility--solving this with packaging would preclude it.

cassidyjames, 2018-01-10:
This is a duplicate of #17

closed 2018-01-10 by cassidyjames

quequotion, 2018-01-11:
oh! sorry about that.