elementary/settings-bluetooth #30: Support authentication code

danrabbit, 2018-03-06:
It seems this is supported by Blueman, so we should support it as well

tintou, 2018-03-06:
I think that this should be moved to the Indicator. It makes sense to me that the default Agent would be registered by the indicator as it is an always running process and can just throw a notification for yes/no pairing

danrabbit, 2018-03-06:
I don't think a notification is good here because the information is only relevant during the time that I'm trying to pair. So if the notification goes away before I pair or it sticks around in the notification indicator after I'm paired, it's not useful.

To be clear, I'm pairing from the computer side and not from the phone side so I have Bluetooth settings open while I'm doing this.

I can't seem to find a screenshot of what GNOME does here, but this is what it looks like on macOS: [image]

geovannikun, 2018-03-06:
Bluetooth auth on Gnome

[image]

tintou, 2018-03-06:
Okay yeah you're right, I thought about incoming connections but I forgot that the device is only visible when the plug is open :sweat_smile: