elementary/settings-bluetooth #31: Do something smart about bluetooth devices with no name

danrabbit, 2018-03-06:
I'm not sure about Android, but it looks like iOS doesn't even list these devices. I'm wondering if we should either also not show them or at least sort them below devices with names.

As it is, I have a big list of nameless devices sorted above and obscuring the devices I actually care about pairing with

tintou, 2018-03-06:
Maybe we should just deprioritize them ?

danrabbit, 2018-03-06:
Yeah I think at the least they should be sorted last, but I wonder if it's even valuable to show. I have no idea what all these devices are but if another major platform is just not even showing them, I wonder if it's important for us to

cassidyjames, 2018-03-06:
Checking devices between my phone and computer… I see several unnamed devices on my computer, but a few devices on my phone that aren't listed on the computer. Is it possible getting the name timed out on the computer, or something?

I think before we know more, deprioritizing them (and maybe naming them something more friendly like "Unknown (MAC:ADDR:IN:DIM:LABEL)" would be useful?

danrabbit, 2018-03-06:
In Slack, Corentin brought up the point that this can be useful for people developing embedded devices etc

PR #43 (2018-05-06): Sort devices with names above devices without
Fixes #31
closed 2018-05-06 by tintou