elementary/settings-bluetooth #35: [Juno] Adapter not discoverable after power on.

jeremypw, 2018-04-01:
On Loki, powering on bluetooth using the wingpanel indicator makes the adapter discoverable.  However, on Juno the adapter is not discoverable and there is (apparently) no way of making it so.

It is recognized that this could be an upstream or driver problem.

It would be nice if discoverability were controllable independently of power on/off.

danrabbit, 2018-04-01:
The device should be discoverable whenever you have the switchboard plug open. Can you confirm that this isn't the case?

jeremypw, 2018-04-01:
Ah, I see - yes it becomes discoverable now when a patched plug I was working on is opened. It becomes undiscoverable again when it is closed. I'll check again with the master plug.  Is this different from Loki?

jeremypw, 2018-04-01:
Same with master.  Is this something a user is expected to know?

danrabbit, 2018-04-01:
I'm pretty sure this is done to save battery. AFAIK, Android, iOS, and macOS all require you to go to the Bluetooth settings page for your device to be discoverable.

jeremypw, 2018-04-01:
OK, I guess I am just ignorant then ; -}

By the way, the adapter becomes undiscoverable if the plug is open for more than 3 minutes (default discoverablity timeout).  My patched version updates the description accordingly.

It might be worth reconsidering the wording of the option "Bluetooth settings" in the bluetooth indicator.  e,g, "Manage Connections"

jeremypw, 2018-04-01:
The bluetooth plug seems to be working better than it was a couple of days ago - I am wondering whether something has been updated upstream ...

Anyway I'll close this now.

closed 2018-04-01 by jeremypw

danrabbit, 2018-04-01:
Ah hm. Yeah we should probably automatically turn discovery back on while the plug is open. Can you open an issue for that?

I also agree about the label in the indicator. It would probably be more discoverable and more accurate since there are not really settings here