elementary/settings-bluetooth #48: Automatically connect after pairing.

arshubham, 2018-07-04:
I don't know if this is intended, but in juno beta when a new device is discovered **Pair** button is showed. On clicking it the button changes to **Disconnect** and subheading to **Disconnected** but only momentarily. The button then shows **Connect** and subheading shows **Paired**. So, I have to click connect again to actually connect my Bluetooth headset.

In summary, it takes two clicked and a wait in between to connect a headset in elementary, while on gnome 3.28 on 18.04 it takes only one click.

Bluetooth Chipset: rtl8723be
Bluetooth Headset: iBall Powebeats.

peteruithoven, 2018-09-26:
Same happens on Dell XPS 15 (9550) with a Atheros Qualcomm Killer 1535 802.11ac 2x2 when pairing with:
- Anker SoundBuds Slim
- Bose QuietControl 30
- UE Boom 2

This issue doesn't occur with my *Logitech M720 Triathlon* bluetooth mouse. Maybe it's an audio device specific issue?

peteruithoven, 2018-10-10:
@arshubham arshubham do you experience the same behaviour when you turn on previously connected audio devices? Or is this something you only see after pairing?

arshubham, 2018-10-10:
I experience this only after pairing. If the device is already paired, I just need to press connect once and it will be connected. However it is not as smooth as on windows. When connecting a already paired device it is connected and workes fine, however after a few minutes my headphone says, `pairing failed`. The latter seems to be a bluez problem as it affects all distros.

peteruithoven, 2018-10-10:
@arshubham But what happens when you turn on a previously connected device? (And don't do anything on your computer) You don't see it momentarily connect? (like you see after pairing)

arshubham, 2018-10-10:
Ok, it doesn't connect automatically if it is already paired, never did on linux. No output in bluetoothctl. just shows devices as paired

PR #58 (2018-10-11): Auto trust after pairing
See: #57 

If I trust the device before pairing it in didn't connect in the high fidelity profile a couple of times for some reason.

Sadly enough this doesn't fix #48.