elementary/settings-bluetooth #52: No padding left and right of settings icon

peteruithoven, 2018-09-26:
This is a visual nitpick. There is no padding to the left and right of the settings (...) button. This is a bit ugly when the row is highlighted. 
(It's height and right margin is also inconsistent with the Disconnect button.)
[image]

peteruithoven, 2018-09-26:
What it looks like if I use a `Gtk.Button` instead of `Gtk.LinkButton` and remove the custom right margin. I guess the downside of this approach is more visual noise. 
[image]

danrabbit, 2022-09-07:
This is a quirk of Gtk.LinkButton in the stylesheet. It's been fixed in Gtk 4, not sure about Gtk3