elementary/settings-bluetooth #55: Discover devices dialog not adapting to number of found devices

thomorl, 2018-10-09:
The "Discover Devices" dialog's list of all detected devices is not increasing or decreasing it's size when new devices are found/devices are lost. The new entries appear in the list, but can't be seen unless the window size is increased manually, making it look like the Bluetooth is not working.

Screenshots:

[image]
**1** Default appearance (note the few visible pixels of the found device icon)

[image]
**2** Manually resized

peteruithoven, 2018-10-09:
The Bluetooth System Settings have had a total redesign, so this separate discovery window for example is no longer necessary. 
[image]

I'm afraid that this however won't be backported to Loki.

peteruithoven, 2018-10-09:
I now see it has also been reported before: https://github.com/elementary/switchboard-plug-bluetooth/issues/15

closed 2018-10-09 by peteruithoven