elementary/settings-bluetooth #57: Auto trust?

peteruithoven, 2018-10-11:
I think I figured out why my audio devices, when I turn them on, seem to automatically connect, but then disconnect again. Apparently this had something to do with the `trust` setting of the device. When I manually trust the device using `$ bluetoothctl` it suddenly does autoconnect when I turn it on. 

Shouldn't we automatically trust devices after pairing them?

Workaround:
1. Run: `$ bluetoothctl`
2. Run: `paired-devices`
3. Copy the ID of the audio device that doesn't auto connect
4. Run: `trust {id}`

More info: https://wiki.archlinux.org/index.php/Bluetooth_headset#Setting_up_auto_connection
Also see: https://github.com/elementary/switchboard-plug-bluetooth/issues/5#issuecomment-424783309

sempasha, 2020-02-13:
Despite autotrust should work, I still experience this issue with all new paired devices (headsets).
```
$ journalctl -f
<date> <user> bluetoothd[<pid>]: Authentication attempt without agent
<date> <user> bluetoothd[<pid>]: Headset Voice gateway rejected <ID>: org.bluez.Error.Rejected
```
After doing these steps there are no more device connection faults:
1. Run: `$ bluetoothctl`
2. Run: `paired-devices`
3. Copy the ID of the audio device that doesn't auto connect
4. Run: `trust {id}`

peteruithoven, 2020-02-13:
I can confirm it doesn't help with headset like devices. It does work for mouse and keyboard.

sempasha, 2020-02-13:
In my case it helps with two bluetooth headsets.