elementary/settings-bluetooth #62: Use disabled icon when bluetooth is disabled

danrabbit, 2018-11-01:
[image]

joshnroy, 2018-11-20:
Hi DanRabit (and anyone else this may pertain to).

I'm a college student and haven't solved many bugs on open source projects before, but found elementary recently, really like it, and would like to try to contribute. I'll try fixing this and see if it works!

Thanks,
Josh

joshnroy, 2018-11-20:
Update: I found the place to change the code, and can now change the icon when bluetooth is on/off. I'm not sure if there's a general list of icons that includes the one you show in your screenshot? I'll look for it

joshnroy, 2018-11-22:
Just pulled the new icon set, and it uses the correct icons now

danrabbit, 2018-11-24:
hey @joshnroy sorry, I submitted a fix for this quite a while ago and forgot to link the issue. I've fixed the link, and the PR has more information about the missing icon and why it's not merged yet

joshnroy, 2018-11-25:
Sounds good. Thanks @danrabbit!

PR #64 (2018-12-06): MainView: Use disabled icon when disabled
Fixes #62 

don't merge this before we get an icons release with https://github.com/elementary/icons/pull/711 or we'll get a gross GNOME fallback icon
closed 2018-12-06 by danrabbit