elementary/settings-bluetooth #74: Apple wireless keyboard can't connect.

Qmo37, 2019-02-04:
Before installed elementary OS, I could connected my apple wireless keyboard to my iMac 2010 mid.
but after I installed OS, I couldn't connect to my iMac, sorry I'm new to Linux.
BTW I've tried this https://support.system76.com/articles/bluetooth/

<bountysource-plugin>

---
Want to back this issue? **[Post a bounty on it!](https://www.bountysource.com/issues/69569379-apple-wireless-keyboard-can-t-connect?utm_campaign=plugin&utm_content=tracker%2F59641246&utm_medium=issues&utm_source=github)** We accept bounties via [Bountysource](https://www.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F59641246&utm_medium=issues&utm_source=github).
</bountysource-plugin>

Qmo37, 2019-02-05:
Thanks for your suggestion.
Somehow, I connected successfully yesterday in pairing mode like you said, but the keyboard didn't work normally, only some parts of keyboard could type and only came out with numbers. No matter how I disconnect and reconnect, it still only numbers.

danrabbit, 2019-04-06:
Closing since this sounds like some kind of hardware failure

closed 2019-04-06 by danrabbit