elementary/settings-bluetooth #87: Some visual issues on the bluetooth settings page

Fatih20, 2019-09-13:
[image]
I don't think the gray rectangle around the 3 dots that is used to configure the device is supposed to exist.

danrabbit, 2019-09-13:
This is a keyboard focus indicator. It is intentional

closed 2019-09-13 by danrabbit

Fatih20, 2019-09-13:
I think this could be made to look better. The edge of the rectangle is too close to the dots make it doesn't look aesthetically pleasing and makes it look like an unintentional mistake.

4jNsY6fCVqZv, 2019-11-04:
I also have the feeling that the visual solution is not as appealing and self-explanatory for me so far.

My scenario:
I have exactly one device that is connected via Bluetooth. In addition, I never use the keyboard for navigation. Nevertheless, I see the grey bar and the grey dots as default. However, when I use the keyboard to navigate, the bars and dots turn blue. I understand the latter because it indicates that the keyboard is in play. What the grey bar and the grey dots want to tell me is unfortunately not clear to me, especially because there are no different elements to choose from.

Could you please explain the concept behind it?

Maybe there is also a visual solution that is more appealing and has the same functionality? For example, if the dots wouldn't stand out like that or the square box around them disappears, things look different.