elementary/settings-bluetooth #88: Add ability to select default bluetooth audio output mode

nixolas1, 2019-09-26:
When connecting my bluetooth headphones, it is always set to Headset mode, which makes it impossible to listen to music. One has to install `blueman` and choose the output as Audio Sink (H2DP) to get normal sound.

**Describe the solution you'd like**
A dropdown with "Default output-mode", which changes the mode on the device, and remembers it the next time it is connected.

Draterko, 2019-10-12:
Switch "Audio Profile" in context menu in "blueman", should be enough for automatically saved selected option per session, at least for me. 
https://i.imgur.com/T2tA8Na.jpg

nixolas1, 2019-10-15:
Yeah, it was enough before. But now with Sony WH-1000XM3 it switches to headset each time it connects, and switching audio profile dosn't work without disconnecting and connecting in the correct mode. So maybe it is device-specific, but still a problem.