elementary/settings-bluetooth #97: [UI] Number code is strangely horizontally aligned

megatux, 2019-11-04:
## Prerequisites
- [x ] I have searched open and closed issues for duplicates.

## Describe the bug

The "code" text alignment looks a bit off to the right, is not centered in the dialog or clear why is there. Probably centered is a better option, as seen on other desktop environments.
[image]

cassidyjames, 2019-11-04:
I think we played with this when prototyping the dialog but couldn't decide what felt the best. It _should_ be centered in the right-side content area, but with wrapping text up above and no clear bounding lines, I agree it feels a little haphazardly placed.

danrabbit, 2019-11-04:
Yeah, we could fiddle with adding extra margin to account for the icon and column spacing, but it is technically centered in its container.

Another option is to left align instead of center. So it would be left aligned against the other text in the dialog