elementary/settings-bluetooth #99: Bluetooth issue

FilipVlcek, 2019-11-16:
Hello. I've got problem with bluetooth. My notebook can not connect via bluetooth to my wireless headphones. When I was trying it absolutely first time it connected. But now it does not and tells me cannoet be connected... I have visibility on and I have tried to reinstall bluetooth adapter but does not work.

FilipVlcek, 2019-11-16:
And I have got a video for you. But do not know how to send it here

kirk-chan, 2019-12-07:
Hello,

I received the same problem and fixed it via:

1. Open terminal
2. _bluetoothctl_
3. Connect to target bluetooth device
4. Trace logs for MAC address
5. _remove <mac_address>_
6. Pair the device again

I believe the pairing may have been corrupted. So this should rebuild the Bluetooth pairing connection.

Cheers!