elementary/settings-bluetooth #103: "Unable to Connect" to paired bluetooth mice

panicsteve, 2019-12-31:
## Prerequisites
- [X] I have searched open and closed issues for duplicates.

## Describe the bug
I have been unable to connect to either of two different bluetooth mice, one Apple branded and one Microsoft branded.

The mice are discovered during the BT scan, and appear to pair successfully, however their status immediately changes to "Unable to Connect".

Additionally, there appears to be no way to unpair a paired device from the GUI.

### To Reproduce

1. Go to Bluetooth pane of settings
2. Place BT mouse in pairing mode
3. Wait for mouse to be discovered by OS
4. Click pair

### Expected behavior

OS pairs with mouse, connects, and becomes functional.

Paired device can be unpaired at a later time from the Bluetooth settings GUI.

### Screenshots or screen recordings

[image]

### Logs

Happy to provide logs -- please indicate which log files are required and where they are stored.

## Platform Information

- [ ] I'm using the latest version from git that I've manually compiled
- [X] I'm using the latest released stable version

[image]

## Additional context

jeremypw, 2019-12-31:
I have noticed this before - I usually have to reconnect using the wingpanel indicator, which seems to work OK.

janxkoci, 2020-02-21:
I don't know if it's exactly the same issue, but me and some other users have problem while connecting to paired speakers / earphones (see [here](https://elementaryos.stackexchange.com/questions/21788/bluetooth-headset-can-no-longer-be-connected)).

The problem seems to be caused by an update to the package `libpulse0`, which behaves very strangely. The new version (`1:11.1-1ubuntu7.5`) for some reason depends on `elementary-desktop`, `gala`, and other packages, so downgrading it requires reinstallation of the desktop. However it seems that this downgrade fixes the bluetooth issue.

Here are some outputs:

```bash
$ apt install --dry-run libpulse0=1:11.1-1ubuntu7.4
INFO: Toto je pouze simulace!
      apt vyžaduje pro skutečný běh rootovská oprávnění.
      Mějte také na paměti, že je vypnuto zamykání, tudíž
      tyto výsledky nemusí mít s realitou nic společného!
Načítají se seznamy balíků… Hotovo
Vytváří se strom závislostí       
Načítají se stavové informace… Hotovo
Následující balíky byly nainstalovány automaticky a již nejsou potřeba:
  io.elementary.cerbere libsnapd-glib1 libwebrtc-audio-processing1
  linux-headers-4.15.0-74 linux-headers-4.15.0-74-generic
  linux-image-4.15.0-74-generic linux-modules-4.15.0-74-generic
  linux-modules-extra-4.15.0-74-generic nautilus-data rtkit
Pro jejich odstranění použijte „apt autoremove“.
Navrhované balíky:
  pulseaudio
Následující balíky budou ODSTRANĚNY:
  elementary-desktop gala gnome-settings-daemon libcanberra-pulse
  libpulse-mainloop-glib0 libpulsedsp pantheon-shell pantheon-xsession-settings
  pulseaudio pulseaudio-utils switchboard-plug-notifications
  switchboard-plug-pantheon-shell switchboard-plug-sound
  wingpanel-indicator-sound
Následující balíky budou DEGRADOVÁNY:
  libpulse0
0 aktualizováno, 0 nově instalováno, 1 degradováno, 14 k odstranění a 0 neaktualizováno.
Remv elementary-desktop [1.468+r534~ubuntu5.0.1]
Remv pantheon-xsession-settings [5.0.3-0~116~ubuntu5.0.1] [pantheon-shell:amd64 ]
Remv pantheon-shell [1.468+r534~ubuntu5.0.1]
Remv switchboard-plug-pantheon-shell [2.8.2+r909+pkg897~ubuntu5.0.1]
Remv gala [3.2.0~r902+pkg49~ubuntu5.1.1] [switchboard-plug-notifications:amd64 ]
Remv gnome-settings-daemon [3.28.1-0ubuntu1.3+elementary4~ubuntu5.0.1] [switchboard-plug-notifications:amd64 ]
Remv libcanberra-pulse [0.30-5ubuntu1] [switchboard-plug-notifications:amd64 ]
Remv wingpanel-indicator-sound [2.1.4+r281+pkg19~ubuntu5.0.1] [switchboard-plug-notifications:amd64 ]
Remv switchboard-plug-sound [2.2.2+r421+pkg13~ubuntu5.0.1] [switchboard-plug-notifications:amd64 ]
Remv libpulse-mainloop-glib0 [1:11.1-1ubuntu7.5] [switchboard-plug-notifications:amd64 ]
Remv pulseaudio [1:11.1-1ubuntu7.5] [switchboard-plug-notifications:amd64 ]
Remv pulseaudio-utils [1:11.1-1ubuntu7.5] [switchboard-plug-notifications:amd64 ]
Remv libpulsedsp [1:11.1-1ubuntu7.5] [switchboard-plug-notifications:amd64 ]
Remv switchboard-plug-notifications [2.1.5+r322~ubuntu5.0.1]
Inst libpulse0 [1:11.1-1ubuntu7.5] (1:11.1-1ubuntu7.4 Ubuntu:18.04/bionic-updates [amd64])
Conf libpulse0 (1:11.1-1ubuntu7.4 Ubuntu:18.04/bionic-updates [amd64])
```
One other weird thing in the dry run is that it mentions `io.elementary.cerbere` as no longer needed. I learned just a few days ago that this package's job is to relaunch `wingpanel` after crash, and now it seems to be removed.

Also note here in `apt show -a` that the new version 7.5 lists tasks, that are not listed for the other version.

```bash
$ apt show -a libpulse0
Package: libpulse0
Version: 1:11.1-1ubuntu7.5
Status: install ok installed
Priority: optional
Section: libs
Source: pulseaudio
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Pulseaudio maintenance team <pkg-pulseaudio-devel@lists.alioth.debian.org>
Installed-Size: 1 023 kB
Depends: libasyncns0 (>= 0.3), libc6 (>= 2.27), libdbus-1-3 (>= 1.9.14), libsndfile1 (>= 1.0.20), libsystemd0, libwrap0 (>= 7.6-4~), libxcb1
Suggests: pulseaudio
Breaks: pavucontrol (<< 0.9.8)
Homepage: http://www.pulseaudio.org
Download-Size: neznámá
APT-Manual-Installed: no
[... 36 lines cut ...]
Version: 1:11.1-1ubuntu7
Priority: optional
Section: libs
Source: pulseaudio
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Pulseaudio maintenance team <pkg-pulseaudio-devel@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 1 022 kB
Depends: libasyncns0 (>= 0.3), libc6 (>= 2.27), libdbus-1-3 (>= 1.9.14), libsndfile1 (>= 1.0.20), libsystemd0, libwrap0 (>= 7.6-4~), libxcb1
Suggests: pulseaudio
Breaks: pavucontrol (<< 0.9.8)
Homepage: http://www.pulseaudio.org
Task: ubuntu-desktop, kubuntu-desktop, xubuntu-core, xubuntu-desktop, lubuntu-desktop-share, lubuntu-gtk-desktop, lubuntu-desktop, lubuntu-qt-core, lubuntu-qt-desktop, lubuntu-qt-desktop, ubuntustudio-video, ubuntustudio-audio, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop
Supported: 5y
Download-Size: 266 kB
APT-Sources: http://cz.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
Description: PulseAudio client libraries
 PulseAudio, previously known as Polypaudio, is a sound server for POSIX and
 WIN32 systems. It is a drop in replacement for the ESD sound server with
 much better latency, mixing/re-sampling quality and overall architecture.
 .
 Client libraries used by applications that access a PulseAudio sound server
 via PulseAudio's native interface.
```

janxkoci, 2020-02-21:
Based on the info above, it is possible that there is a need to patch some upstream package, so in such case consider moving to https://github.com/elementary/os-patches.

janxkoci, 2020-02-24:
So it's probably more complex, because I have the same package on another laptop and there it connects fine.

> I usually have to reconnect using the wingpanel indicator, which seems to work OK.

This doesn't work for me btw. Both switchboard and indicator are consistently unable to connect to the earbuds.