elementary/settings-bluetooth #108: Bluetooth Setting and indicator missing on fresh install EOS Hera

alfianhanafi, 2020-01-27:
I just install Elementary OS Hera and i dont find Bluetooth setting and BT Indicator

**Setting**
[image]

**Indicator**
[image]

Please Help

tintou, 2020-01-27:
Do you have a bluetooth adapter included in your computer, is it disabled using an hardware switch?

alfianhanafi, 2020-01-27:
> Do you have a bluetooth adapter included in your computer, is it disabled using an hardware switch?

I think not, i mean my wifi is fine.
it has the same button for wifi and Bluetooth (ALT+F2).

im using Asus laptop BTW
when i install Juno it's working properly

alfianhanafi, 2020-01-27:
my BT status

[image]

closed 2020-01-27 by alfianhanafi