elementary/settings-bluetooth #131: Buttom to unpair bluetooth devices

gersonjunior87, 2020-07-02:
I looked for a button to unpair a device that was not connecting correctly and I couldn't find it. I removed it via terminal. It would be interesting to have an unpair button in the device's bluetooth settings.

danrabbit, 2020-07-02:
There is a “-“ button on the toolbar for unpairing. Admittedly it is very understated. This isn’t the first we’ve heard that it is unclear.

rafaelcorsi, 2020-07-11:
@danrabbit working on this, i`ve two solutions:

1. Improve visibility on the actual button:

[image]

2. add a individual button to each connection:

[image]

gersonjunior87, 2020-07-11:
Wow, I loved the second choice (but the footer button need be removed, right?)

danrabbit, 2020-07-12:
@rafaelcorsi Thanks for looking into this! I agree that option 2  (with buttons on each row) is probably the most obvious. And like @gersonjunior87 mentioned, we can then remove the button from the toolbar below.

rafaelcorsi, 2020-07-13:
@danrabbit ive implemented de seccond option, but I think I have a better solution. 

The idea is to add popover button to all bluetooth devices (not to only that the have a configuration), and on this new window add the option to Forget and Settings plus some information as: 

- Device address (actually is a tooltip)
- Date of last time user connect to device (I dont know if bluz has this information)
- Baterry (is this redundant to the wingpanel indicator?)
- Forget / Settings buttons ( the settings button only appear on some devices)

[image]

I was inspired on the wifi settings:

[image]

I would like to know if you think it is worth spending more energy on this other idea? Or I send or PR option 2?

scottjenson, 2020-07-19:
If you don't mind a new comer chiming in... I'm not up to speed on your style guide but the unpair button, while dangerous isn't very common. We should be making it more visible yes, but having all of those red buttons is a bit in-the-users face. It might help to have a clear, but more understated option.

As to the pop up menu, that clearly offers lots more help but it suffers from being *just* that much more hidden. It risks being as hard to find as the original "-" button. Here is a quick mock up but likely I said, I'm new here so if there is a better style guide to follow (e.g. having borders on all your buttons) please let me know
[image]

danrabbit, 2020-07-19:
@rafaelcorsi I'm not sure I'm convinced we need a whole menu here, but in any case it would probably be better to save that proposal for another pull request.

I would be interested in @scottjenson's solution of using an image button here. I agree that this is probably not a very common action and so deemphasizing it would be good

If you do decide to go with an image button, remember that image buttons need tooltips :)

PR #133 (2020-09-22): Add a forget button to each device
closes #131 . 

Add individual forget button to each paired bluetooth device.

On this PR:

- remove the forget button from the toolbar 
- add forget button to each paired element 
- create a new state to know the difference between: Paired -> not connected / unPaired -> not connected.
- a little code refactoring: I created a function to configure the uri setting button 

I decided to use an Image + EventBox to avoid the icon to be highlighted when the device row is selected, as actually occurs with the device settings (I will try to fix and send in a new PR) [... cut]
closed 2020-09-22 by danrabbit