elementary/settings-bluetooth #172: Bluetooth pairing request dialog does not close on pairing error

AI-ien, 2021-03-18:
## What Happened
I'm trying to pair Logitech K375s keyboard, but it fails. Pop-up window with passkey appears, but none of the buttons work, neither close nor pair. Keyboard keeps sending pins filling in new windows with request for pairing until device is turned off. In logs I see some HID errors, but I have not pinned down yet what exactly is happening.

## Expected Behavior
On bluetooth pairing error pin request window should destroy itself (optionally showing error window).
Also window should be able close on close button click, that does not happen.

## Steps to Reproduce
Try to pair device, that request passkey confirmation, but fails to pair.

### Logs
I didn't find any errors related to window closing (its possible that I don't know where to look). I found only bluetooth errors, but it is not about that so.

```bash
$ sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-03-18 12:01:07 EET; 19min ago
     Docs: man:bluetoothd(8)
 Main PID: 931 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─931 /usr/lib/bluetooth/bluetoothd

mar 18 12:01:39 kerrigan bluetoothd[931]: Error reading PNP_ID value: Request attribute has encountered an unlikely error
mar 18 12:01:39 kerrigan bluetoothd[931]: HID Information read failed: Request attribute has encountered an unlikely error
mar 18 12:01:39 kerrigan bluetoothd[931]: Report Map read failed: Request attribute has encountered an unlikely error
mar 18 12:01:39 kerrigan bluetoothd[931]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely 
mar 18 12:01:39 kerrigan bluetoothd[931]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely 
mar 18 12:01:39 kerrigan bluetoothd[931]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely 
mar 18 12:01:39 kerrigan bluetoothd[931]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely 
mar 18 12:01:39 kerrigan bluetoothd[931]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely 
mar 18 12:01:39 kerrigan bluetoothd[931]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely 
mar 18 12:01:39 kerrigan bluetoothd[931]: Protocol Mode characteristic read failed: Request attribute has encountered an unlikely
```

## Platform Information
```bash
         eeeeeeeeeeeeeeeee            arnis@kerrigan 
      eeeeeeeeeeeeeeeeeeeeeee         -------------- 
    eeeee  eeeeeeeeeeee   eeeee       OS: elementary OS 5.1.7 Hera x86_64 
  eeee   eeeee       eee     eeee     Host: Aspire VX5-591G V1.08 
 eeee   eeee          eee     eeee    Kernel: 5.11.7-051107-generic 
eee    eee            eee       eee   Uptime: 29 mins 
eee   eee            eee        eee   Packages: 2376 
ee    eee           eeee       eeee   Shell: bash 4.4.20 
ee    eee         eeeee      eeeeee   Resolution: 1920x1080, 2560x1440 
ee    eee       eeeee      eeeee ee   DE: Pantheon 
eee   eeee   eeeeee      eeeee  eee   WM: Mutter(Gala) 
eee    eeeeeeeeee     eeeeee    eee   Terminal: io.elementary.t 
 eeeeeeeeeeeeeeeeeeeeeeee    eeeee    CPU: Intel i7-7700HQ (8) @ 3.800GHz 
  eeeeeeee eeeeeeeeeeee      eeee     GPU: NVIDIA GeForce GTX 1050 Ti Mobile 
    eeeee                 eeeee       GPU: Intel Integrated Graphics 
      eeeeeee         eeeeeee         Memory: 2321MiB / 15872MiB 
         eeeeeeeeeeeeeeeee 

```
[image]

AI-ien, 2021-03-18:
### Found solution and workaround

After some tinkering I figured out what is going on. When you are pairing Logitech keyboards (possibly others as well) it actually asks to enter pairing pin **on the keyboard**, but dialog window does not say that. Text actually is "Make sure code displayed on <keyboard name> matches the one below." with two buttons to press `Cancel` and `Pair`. Well, the truth is keyboards does not have displays.

My guess is that this kind of behavior is not implemented in this dialog, that results in unable to pair.

Workaround for this is simply type code on the keyboard followed by `Enter` when pairing window appears and ignore buttons. Also after successful pairing `Pairing window` will not close so use `Esc` to close it.
Another option is to use gnome control center as it displays proper text.

peteruithoven, 2021-03-23:
Those confirm Bluetooth Passkey windows used to automatically disappear when the pairing was successful. I have the feeling they stopped disappearing only recently, maybe after: https://github.com/elementary/switchboard-plug-bluetooth/releases/tag/2.3.5
Changes: https://github.com/elementary/switchboard-plug-bluetooth/compare/2.3.4...2.3.5

AI-ien, 2021-03-23:
For me they do disappear when successful. Problem is when not successful and when pairing keyboards, that request to enter pin on keyboard.

peteruithoven, 2021-03-23:
But you said:
> Also after successful pairing Pairing window will not close so use Esc to close it.

peteruithoven, 2021-03-23:
The following bit seems relevant for the text that isn't really appropriate:
https://github.com/elementary/switchboard-plug-bluetooth/blob/master/src/PairDialog.vala#L93-L119

jaw6565, 2021-08-12:
I have a the problem of when the Bluetooth keyboard is paired the pairing notice with the coad stays on the screen, the only way to remove it is to reboot the laptop.

danirabbit, 2025-06-18:
Issues with the incorrect instructions, the dialog not closing automatically, and buttons not working have all been fixed. The issue with duplicate dialogs is being tracked here: https://github.com/elementary/switchboard-plug-bluetooth/issues/186

closed 2025-06-18 by danirabbit