elementary/settings-bluetooth #174: Bluetooth activates - after suspended

johnvenishmael-12, 2021-04-19:
My bluetooth is deactivated. When I suspended my laptop for some reason and went back to operation the bluetooth is activated unlike when I restart my unit. When restart, bluetooth is activated then it will deactivated after I login.

vjr, 2021-04-20:
What is the model of wifi and Bluetooth chipset do you have on your laptop?

vjr, 2021-04-20:
Potentially fixed by https://github.com/elementary/wingpanel-indicator-bluetooth/pull/143

johnvenishmael-12, 2021-04-22:
> Potentially fixed by elementary/wingpanel-indicator-bluetooth#143

Hi am new to linux and regards to chipset im not sure about hardware info,
app.

please see if attached photo for my laptop device / PCI device info

shawn1thompson, 2021-08-05:
This seems to be fixed for me due to an update yesterday I believe.

vjr, 2021-08-05:
> This seems to be fixed for me due to an update yesterday I believe.

Are you running the Odin elementary OS 6.0 beta?

Would you mind mentioning what wifi+bluetooth chipset model you have in your computer?

johnvenishmael-12, 2021-08-05:
Hi

i have no issues with Hera elementary OS. i think its just normal when open
the computer to activate all network. thank you

shawn1thompson, 2021-08-05:
I am running Odin, Beta 2. Here's `lspci -nnk | grep -iA2 net`:

```
02:00.0 Network controller [0280]: Intel Corporation Wireless 8260 [8086:24f3] (rev 3a)
	Subsystem: Intel Corporation Wireless 8260 [8086:0110]
	Kernel driver in use: iwlwifi
```

shawn1thompson, 2021-08-11:
I spoke too soon. I installed the stable release yesterday and it's back, if it ever left.

o-alquimista, 2021-08-12:
> What is the model of wifi and Bluetooth chipset do you have on your laptop?

Dell Inspiron 15 3583 - [see specs here](https://www.dell.com/support/home/en-us/product-support/servicetag/0-cGZHZExQSms2RFJvY1RSRVZ6Q3ZMQT090/overview)

```
Intel 9560/9462 wireless driver (iwlwifi)
```

I can reproduce this issue.

theuser18, 2021-08-19:
This issue affects me as well.

zhrexl, 2021-09-19:
Can this be related with the fact that, by default, bluetooth is turned on when we open bluetooth settings?
[image]

o-alquimista, 2022-01-09:
> Dell Inspiron 15 3583 - [see specs here](https://www.dell.com/support/home/en-us/product-support/servicetag/0-cGZHZExQSms2RFJvY1RSRVZ6Q3ZMQT090/overview)
>
> ```
> Intel 9560/9462 wireless driver [...]

I can still reproduce it on elementary 6.1 (Jólnir)

o-alquimista, 2022-07-09:
@vjr Hi, I've added the requested information.

vjr, 2022-07-10:
Hi ... I'll try to find some time to further look into this issue although if I can't reproduce it I might not be able make much progress. Stay tuned.