elementary/settings-bluetooth #186: Issues with pairing Logitech K380 Bluetooth Keyboard

taulant, 2021-09-07:
## What I am expecting to happen
1. Open Bluetooth preferences
2. Add keyboard
3. Authenticate by typing the code that is displayed in the Bluetooth authentication process
4. Finish authorizing the keyboard
5. Keyboard is added to the saved devices list

## What is happening
1. Open Bluetooth preferences
2. Add keyboard
3. Authenticate by typing the code that is displayed in the Bluetooth authentication process
4. **While trying to authenticate the keyboard a new authentication window panel is opened for every authentication character entered through the keyboard. This fills up the screen with authentication windows that you cant close thus freezing the preferences panel.**

Sometimes the authentication will succeed (although the preferences panel will be frozen) the OS has trouble re-pairing the keyboard after a restart.

This is reproducible for me 100%.

I think it could be reproducible with other keyboards that require entering a string for authentication.

Thomas-O, 2021-09-07:
I have exactly the same issue with the K380 keyboard and a new install of Odin. The keyboard can be paired and will be full functioning. However, Elementary will not connect to the keyboard after a reboot. Though it will connect perfectly to my Logitech M350 bluetooth mouse. Which doesn't require an authentication code to pair. So I can at least log-in with the on-screen keyboard and then delete the keyboard in the bluetooth settings before pairing the keyboard again. This works most of the time but it's becoming a bit of a pain. I may need to buy / borrow a usb keyboard.

taulant, 2021-09-11:
@Thomas-O I hear you. Having a similar issue. Even if I somehow get it to connect after restarting I have to go through it all over again.

To me it seems that the string input authentication modal has not been fully configured and it is using a fallback modal screen that is used for code confirmation (usually used when pairing a device that has a screen). It is just displaying the authentication code and it is not letting you do any data entry.

Guardian-AI-Dusty, 2021-11-02:
Btw 

- This issue is still present in elementary os 6.0 
- I just hope it gets fixed soon :(
[image]

As you can see it does get connected however the pop up doesn't go away a on reboot the keyboard doesn't connect again

marcinjahn, 2021-11-05:
I have exactly the same issue. The same keyboard (Logitech K380), eOS 6.0.3.
I have a comparison with Fedora 34/35 where the keyboard works flawlessly. Additionally, the pairing process is so much better on Fedora. While typing the code, the characters appear on the screen as dots (as if I was typing in a password). On elementaryOS, the window with the code just pops up for each key press. Then, when I finish typing, and press enter, I have to close all these windows, and then the keyboard works. Until I restart...

It is a bit surprising, because I had this keyboard working perfectly fine on eOS 5.1.7 for months. There were occasional hiccups (randomly all Bluetooth devices were disconnected), but after a restart it was going back to normal.