elementary/settings-bluetooth #187: Send file via Bluetooth

0kevi, 2021-09-23:
Didn't see any issue with the subject. The feature was there in Hera. It's great. Just right-click and send a file to your smartphone. Unfortunately the feature isn't there anymore in Elementary OS 6 (or it is hidden).

Thanks
Steve

jeremypw, 2021-09-23:
You probably need to install this: https://github.com/elementary/gnome-bluetooth-contract.

The use of contracts needs reworking for Odin since they do not work with FlatPak apps but it should work with Files, which is not Flatpak'd yet.

0kevi, 2021-09-23:
Thanks Jeremy, but no..didn't work.

gnome-bluetooth is already the newest version (3.34.3-0ubuntu1).
gnome-bluetooth set to manually installed.

We need to wait for the reworking. There are workarounds, but Bluetooth is elegant.

danirabbit, 2024-05-23:
Closing since bluetooth send/receive was implemented in another release

closed 2024-05-23 by danirabbit