elementary/settings-bluetooth #190: Bluetooth paring with headphone or earbuds

MLTK3, 2021-11-29:
can't believe this is still not working!! just installed your OS, did not sign up to be a beta tester. Don't have the time. This has been in que since July. B&O portal headphones with Surface pro 4, . Apple fit pro, they both show as paired, but no audio. Also show double entry for the bluetooth devices. So effectivlty, Bluetooth does not work....this must be beta software.

ayoungethan, 2022-03-03:
I also want bluetooth to work better in eOS, but it won't unless we provide more constructive support. As it stands this report is just a critical rant and is likely to get ignored. Can you submit a full bug report using the instructions that they carefully provide? There is not enough information yet to be usable and actionable.

danirabbit, 2024-06-11:
I'm not able to test with that specific hardware, but Apple Airpods Pro are playing audio over bluetooth here. There might be an issue specific to the Surface Pro 4, which would be a Linux Kernel issue. There's been quite a few kernel updates since 2021, so I'm going to assume this is fixed. Either way, the bluetooth stack is developed upstream and specific hardware support isn't something we can improve in our codebase

closed 2024-06-11 by danirabbit