elementary/settings-bluetooth #201: bluetooth turns off only from the panel

Cosmosiai, 2023-03-08:


stan-janssen, 2023-03-31:
So the Switchboard plug Bluetooth turns on the Bluetooth radio as soon as you open de plug. kind of understand the reasoning: if you go into that panel, you probably want to do something with Bluetooth, but it might be confusing.

I'm not sure if I'm reading the bug report correctly, but I think there is also a complaint that you *cannot* turn off Bluetooth using the switchboard plug. That is something that I solved in this PR: https://github.com/elementary/wingpanel-indicator-bluetooth/pull/191. It was some weird interaction between Switchboard and Wingpanel. See the video in that PR.

jeremypw, 2023-04-08:
Issue reopened as previous fix is likely to be reverted.

PR #205 (2023-04-08): Fix wingpanel bluetooth indicator settings fight
Fix #201 

This stops the wingpanel indicator overriding the plug and turning Bluetooth back on when switched off in the plug.  The plug no longer changes the wingpanel settings.  Instead it only interacts with the global state and turns the Bluetooth adaptor(s) on and off.  It is left to the wingpanel indicator to pick up the change and make any necessary changes to the UI and settings.

A related PR to the wingpanel will be pushed to completely synchronise the settings, plug and indicator, but this is not required for this PR to work.

An unne [... cut]