elementary/settings-bluetooth #213: Pairing with keyboard issues

GameBoyYeet, 2023-11-12:
### What Happened?

1. I tried to pair my keyboard
2. There's a popup asking if the code matches what's displayed on the other device. However, this popup should prompt the user to type it into their keyboard instead. Nonetheless, the passkey still works if typed into the keyboard.
3. When I proceed to enter the passkey into my keyboard, there's no indication that the keyboard has paired successfully (except for the keyboard working and the device showing up in the Bluetooth list.) Also, I cannot exit the passkey prompt by clicking the buttons, I must press Escape.

### Steps to Reproduce

1. Pair a Bluetooth keyboard
2. Observe the wording of the popup not matching the action
3. Observe that the popup cannot be closed with the buttons

[image]

### Expected Behavior

The popup should prompt the user to type the passkey into their keyboard, and press Enter if needed. Also, the prompt should close if the device pairs successfully. Otherwise, I should be able to close the prompt by clicking on its buttons.

In my eyes, the prompt should look something like this (with the elementary look):
[image]

### OS Version

7.x (Horus)

### Software Version

Latest release (I have run all updates)

GameBoyYeet, 2023-11-12:
Oh, seems like someone already posted about the issue... #200

zeebok, 2023-11-17:
Your issue adds the message language as an issue as well, so I won't close it.

closed 2025-06-18 by danirabbit